platformio/platform-ststm8 #1: Add as platfrom STM8

namoaton, 2018-12-14:
Board  STM8L Discovery
compiler SDCC - Small Device C Compiler

- Arduino Core - https://github.com/tenbaht/sduino

ivankravets, 2018-12-14:
Folks, do you have any board from this list https://github.com/tenbaht/sduino/blob/development/sduino/hardware/sduino/stm8/boards.txt ?

Gei0r, 2018-12-21:
I have one of those extremely cheap STM8S103 breakout boards, which I guess is called stm8sblue in your link.
Although my board has an STM8S103F**2**, which only has 4 KiB of Flash.

I'll be willing and able to help implementing and testing.

[image]

ivankravets, 2018-12-21:
1. Please open PlatformIO IDE Terminal and type
```
pio platform install https://github.com/platformio/platform-ststm8.git
```
2. Restart VSCode and create a new project using PlatformIO IDE > PIO Home > New Project

Does it work?

Gei0r, 2018-12-22:
I'm not using VS Code, I'm running platformio directly from the command line (plain `cmd` on Win10).

Compiling works.

But when uploading, the following error occurs:

    Configuring upload protocol...
    AVAILABLE: serial
    CURRENT: upload_protocol = serial
    Looking for upload port...
    Error: Please specify `upload_port` for environment or use global `--upload-port` option.
    For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
    *** [upload] Explicit exit, status 1

Upload command should be:

    stm8flash.exe -cstlinkv2 -pstm8s103f2 -s flash -w .pioenvs\blue\firmware.ihx

However, this command mysteriously only works about 10 % of the time. But that might be some bad wiring, I'll keep investigating.

Gei0r, 2018-12-22:
I now tried VS Code. Same problem with the upload tool.

Another problem is that when creating a new project, a `main.cpp` file is automatically created, which confuses the build system (SDCC is only a C compiler):

    Compiling .pioenvs\stm8sblue\FrameworkArduino\WInterrupts.c.rel
    In file included from C:\users\a\.platformio\packages\framework-arduinoststm8\cores\sduino/Arduino.h:32:0,
    from src\main.cpp:1:
    C:\users\a\.platformio\packages\framework-arduinoststm8\STM8S_StdPeriph_Driver\inc/stm8s.h:90:3: error: #error "Unsupported Compiler!"
    #error "Unsupported Compiler!"          /* Compiler defines not found */
    ^~~~~
    In file included from C:\users\a\.platformio\packages\framework-arduinoststm8\cores\sduino/Arduino.h:32:0,
    from src\main.cpp:1:
    C:\users\a\.platformio\packages\framework-arduinoststm8\STM8S_StdPeriph_Driver\inc/stm8s.h:2762:11: fatal error: intrinsics.h: No such file or directory

It works when the `main.cpp` is manually renamed to `main.c`.

ivankravets, 2019-01-11:
Resolved in https://github.com/platformio/platform-ststm8/releases/tag/v1.0.0

Please file a new issue if you have any problems.

closed 2019-01-11 by ivankravets