platformio/platform-ststm8 #2: Home: Could not initialize project (STM8S)

ludiazv, 2019-01-26:
PIO Core Call Error: "The next files/directories have been created in /Users/ldiaz/experiments/mcu/STM8_TST\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: spl)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html\nPLATFORM: ST STM8 > ST STM8S103F3 Breakout Board\nHARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)\nOSError: [Errno 2] No such file or directory:\nFile \"/Users/ldiaz/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py\", line 169:\nenv.SConscript(\"$BUILD_SCRIPT\")\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\nreturn _SConscript(self.fs, *files, **subst_kw)\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\nexec _file_ in call_stack[-1].globals\nFile \"/Users/ldiaz/.platformio/platforms/ststm8/builder/main.py\", line 98:\ntarget_elf = env.BuildProgram()\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\nreturn self.method(*nargs, **kwargs)\nFile \"/Users/ldiaz/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py\", line 107:\nenv.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\nreturn self.method(*nargs, **kwargs)\nFile \"/Users/ldiaz/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py\", line 319:\nSConscript(env.GetFrameworkScript(f), exports=\"env\")\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 604:\nreturn method(*args, **kw)\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\nreturn _SConscript(self.fs, *files, **subst_kw)\nFile \"/Users/ldiaz/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\nexec _file_ in call_stack[-1].globals\nFile \"/Users/ldiaz/.platformio/platforms/ststm8/builder/frameworks/spl.py\", line 89:\nsrc_filter=[\"-<*>\"] + [\" +<%s>\" % f for f in get_core_files()]\nFile \"/Users/ldiaz/.platformio/platforms/ststm8/builder/frameworks/spl.py\", line 49:\ncwd=join(FRAMEWORK_DIR, \"Libraries\", \"STM8S_StdPeriph_Driver\", \"inc\")\nFile \"/Users/ldiaz/.platformio/penv/lib/python2.7/site-packages/platformio/util.py\", line 401:\np = subprocess.Popen(*args, **kwargs)\nFile \"/Users/ldiaz/.pyenv/versions/2.7.12/lib/python2.7/subprocess.py\", line 711:\nerrread, errwrite)\nFile \"/Users/ldiaz/.pyenv/versions/2.7.12/lib/python2.7/subprocess.py\", line 1343:\nraise child_exception\n========================== [ERROR] Took 1.29 seconds =========================="

ivankravets, 2019-01-26:
@valeros could you reproduce it?

ivankravets, 2019-01-30:
Thanks! This is a bug. Please update to https://github.com/platformio/platform-ststm8/releases/tag/v1.0.1

ivankravets, 2019-01-31:
See https://bintray.com/beta/#/platformio/dl-platforms/ststm8?tab=overview

Could you try again `pio update`?

ludiazv, 2019-01-31:
It worked. Now the platform is updated.

Thanks

commit 54a6af9f (2019-01-30): Fix "OSError: [Errno 2] No such file or directory" when SPL framework is used // Resolve #2