platformio/platform-ststm8 #4: Build fails with "unsupported compiler"/"intrinsics.h"

adlerweb, 2019-02-10:
The current readme only states to use the platformio.ini to activate ststm8, however I wasn't able to get the target working on both, linux and windows, using stable and development targets.

Log on archlinux using arduino:
```
Processing stm8sblue (platform: https://github.com/platformio/platform-ststm8.git; board: stm8sblue; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 16 compatible libraries
Scanning dependencies...
No dependencies
g++ -o .pioenvs/stm8sblue/src/main.cpp.rel -c -DF_CPU=16000000L -DPLATFORMIO=30604 -DSTM8S_BLUE -DSTM8S103 -DARDUINO_ARCH_STM8 -DARDUINO=10802 -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -Isrc -Iinclude -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/variants/standard src/main.cpp
sdcc -o .pioenvs/stm8sblue/FrameworkArduino/SPI.c.rel -c -mstm8 -DF_CPU=16000000L -DPLATFORMIO=30604 -DSTM8S_BLUE -DSTM8S103 -DARDUINO_ARCH_STM8-DARDUINO=10802 -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/variants/standard /home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/SPI.c
sdcc -o .pioenvs/stm8sblue/FrameworkArduino/WInterrupts.c.rel -c -mstm8 -DF_CPU=16000000L -DPLATFORMIO=30604 -DSTM8S_BLUE -DSTM8S103 -DARDUINO_ARCH_STM8 -DARDUINO=10802 -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/variants/standard /home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/WInterrupts.c
sdcc -o .pioenvs/stm8sblue/FrameworkArduino/WMath.c.rel -c -mstm8 -DF_CPU=16000000L -DPLATFORMIO=30604 -DSTM8S_BLUE -DSTM8S103 -DARDUINO_ARCH_STM8 -DARDUINO=10802 -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino-I/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc -I/home/adlerweb/.platformio/packages/framework-arduinoststm8/variants/standard /home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/WMath.c
In file included from /home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:32,
from src/main.cpp:1:
/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc/stm8s.h:90:3: error: #error "Unsupported Compiler!"
#error "Unsupported Compiler!"          /* Compiler defines not found */
^~~~~
In file included from /home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:32,
from src/main.cpp:1:
/home/adlerweb/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/inc/stm8s.h:2762:11: fatal error: intrinsics.h: No such file or directory

********************************************************************
* Looking for intrinsics.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:intrinsics.h"
* Web  > https://platformio.org/lib/search?query=header:intrinsics.h
*
********************************************************************

#include <intrinsics.h>
^~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/stm8sblue/src/main.cpp.rel] Error 1
/home/adlerweb/.platformio/packages/framework-arduinoststm8/cores/sduino/WInterrupts.c:73: warning 85: in function attachInterrupt unreferenced function argument : 'mode'
=========================================================== [ERROR] Took 1.30 seconds ===========================================================
```

The compiler installed by platformio looks to be `SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ez80_z80/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.8.4 #10766 (Linux)`. A similar error appears when using spl instead of arduino. As far as I can tell the build process uses SDCC but doesn't always define _SDCC_/__SDCC. Manually adding these (build_flags) doesn't do the trick either.

For windows I couldn't get any sane error messages - it seems to randomly fail with either the same message as stated above or "g++ not found".

terragit, 2019-03-20:
I had a similar problem with a newly created project. Unfortunately, I don't have a solution. But after some testing, I got myself the "arduino-fade-all-pins" examples from this repo:
https://github.com/platformio/platform-ststm8
I can compile it and flash to my STM8.

xfdr0805, 2019-05-31:
I had a similar problem with a newly created project.

Nandopolis, 2019-06-07:
I had the same problem with a new project created from PIO Home on vscode
I solved this by renaming the main.cpp file to main.c
hope this helps.

xfdr0805, 2019-06-07:
> I had the same problem with a new project created from PIO Home on vscode
> I solved this by renaming the main.cpp file to main.c
> hope this helps.
me too,but i have a new problem at

mh4nx7net, 2020-12-09:
```
(penv) (researchDev) ☁  SmartVillages-endsec-gps  pio run          
Processing stm8sblack (platform: ststm8; board: stm8sblack; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblack.html
PLATFORM: ST STM8 (1.0.2) > ST STM8S105K4T6 Breakout Board
HARDWARE: STM8S105K4T6 16MHz, 2KB RAM, 16KB Flash
PACKAGES: 
 - framework-arduinoststm8 0.40.181216 
 - tool-stm8binutils 0.230.0 (2.30) 
 - toolchain-sdcc 1.30804.10766 (3.8.4)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/stm8sblack/firmware.elf
ASlink-Warning-No definition of area SSEG
*** [.pio/build/stm8sblack/firmware.elf] Error 1
```

dont help yet :/

valeros, 2021-03-23:
Sduino doesn't support C++ sources. Please use the `.c` extension. 

@mh4nx7net You can try the upstream version of the platform:
```ini
[env:test]
platform = https://github.com/platformio/platform-ststm8.git
framework = ...
board = ...
```

closed 2021-03-23 by valeros