platformio/platform-ststm8 #5: Build failed in vsCode on mac os

Corundex, 2019-02-15:
Just created empty project:

Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 21 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/stm8sblue/src/main.cpp.rel
Archiving .pioenvs/stm8sblue/libFrameworkArduino.lib
Indexing .pioenvs/stm8sblue/libFrameworkArduino.lib
In file included from src/main.cpp:1:
In file included from /Users/abutko/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:23:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/stdlib.h:94:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdlib.h:168:1: error: 'long float' is invalid
long double
^
In file included from src/main.cpp:1:
In file included from /Users/abutko/.platformio/packages/framework-arduinoststm8/cores/sduino/Arduino.h:26:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/math.h:301:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:133:26: error: 'long float' is invalid
extern int __fpclassifyl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:176:47: error: 'long float' is invalid
__header_always_inline int __inline_isfinitel(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:179:44: error: 'long float' is invalid
__header_always_inline int __inline_isinfl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:182:44: error: 'long float' is invalid
__header_always_inline int __inline_isnanl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:185:47: error: 'long float' is invalid
__header_always_inline int __inline_isnormall(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:188:46: error: 'long float' is invalid
__header_always_inline int __inline_signbitl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:196:47: error: 'long float' is invalid
__header_always_inline int __inline_isfinitel(long double __x) {
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:205:44: error: 'long float' is invalid
__header_always_inline int __inline_isinfl(long double __x) {
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:214:44: error: 'long float' is invalid
__header_always_inline int __inline_isnanl(long double __x) {
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:228:46: error: 'long float' is invalid
__header_always_inline int __inline_signbitl(long double __x) {
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:230:9: error: 'long float' is invalid
long double __ld;
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:249:47: error: 'long float' is invalid
__header_always_inline int __inline_isnormall(long double __x) {
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:310:8: error: 'long float' is invalid
extern long double acosl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:310:26: error: 'long float' is invalid
extern long double acosl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:314:8: error: 'long float' is invalid
extern long double asinl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:314:26: error: 'long float' is invalid
extern long double asinl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:318:8: error: 'long float' is invalid
extern long double atanl(long double);
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:318:26: error: 'long float' is invalid
extern long double atanl(long double);
^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
*** [.pioenvs/stm8sblue/src/main.cpp.rel] Error 1
=============================================================================================== [ERROR] Took 0.92 seconds ===============================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

TimotF, 2019-06-03:
I have the exact same issue, did you manage to fix it?

fexeak, 2019-11-28:
Anybody fix it now ?

viktor-buyakov, 2020-03-13:
Any updates ?
Got same error for stm8sblue.

postpersonality, 2020-04-16:
Wow. So, it's broken for more than a year now. That's sad.

postpersonality, 2020-04-16:
https://community.platformio.org/t/begginer-in-platform-io-need-help-with-programming-stm8/9078/4

The solution is to use `main.c` instead of `main.cpp`

valeros, 2021-03-23:
Sduino doesn't support C++ sources. Please use `.c` extension for source files.

closed 2021-03-23 by valeros