platformio/platform-ststm8 #7: [env] upload_protocol not applied correctly / board_config.get("upload.protocol") wrong value

terragit, 2019-03-20:
My platformio.ini (PIO Home 2.0.0, Core 3.6.5, no pending updates)
> [env:stm8sblue]
> platform = ststm8
> framework = arduino
> board = stm8sblue
> 
> upload_protocol = stlinkv2

When trying to upload aka flash (via PIO Verbose Upload) a small blink example to an STM8, it fails with this error:
> AVAILABLE: serial, stlinkv2
> CURRENT: upload_protocol = stlinkv2
> "stm8flash" **-c serial** -p stm8s103?3 -s flash -w .pioenvs/stm8sblue/firmware.ihx
> ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
> 
> No valid programmer specified. Possible values are:
> stlink
> stlinkv2
> stlinkv21
> stlinkv3
> espstlink
> *** [upload] Error 255

As can be seen, stm8flash does use **-c serial** although *CURRENT: upload_protocol = stlinkv2* is printed.
I could not find the error but a workaround:
Changed line 152 of file .platformio/platforms/ststm8/builder/main.py from
> "-c", board_config.get("upload.protocol"),
to 
> "-c", upload_protocol

Hope this description helps.

ludiazv, 2019-03-26:
this is my workaround for stlinkv2:
[env:stm8sblue]
platform = ststm8
board = stm8sblue
framework = spl
upload_protocol = stlinkv2
board_upload.protocol = stlinkv2
board_build.mcu= stm8s103F3
build_flags= --all-callee-saves --debug --stack-auto --fverbose-asm --float-reent --no-peep

terragit, 2019-03-26:
> board_upload.protocol = stlinkv2

That's the solution. Just replace **upload_protocol** by **board_upload.protocol**. 
Thank you, @ludiazv

ludiazv, 2019-03-28:
I think it should be changed to be aligned with other platforms in PIO. I found the solution by trial an error peaking the source code. But also this is not documented.

ivankravets, 2019-03-28:
Thanks for the report! Please re-test with upstream version  https://docs.platformio.org/en/latest/platforms/ststm8.html#stable-and-upstream-versions

closed 2019-03-28 by ivankravets

jay-ramani, 2022-03-28:
> this is my workaround for stlinkv2: [env:stm8sblue] platform = ststm8 board = stm8sblue framework = spl upload_protocol = stlinkv2 board_upload.protocol = stlinkv2 board_build.mcu= stm8s103F3 build_flags= --all-callee-saves --debug --stack-auto --fverbose-asm --float-reent --no-peep

I hit the issue in the most recent version of PlatformIO. Took some searching, but using ```board_upload.protocol``` saved the day. Thank you so much for sharing the solution. I just wish this had been documented.

@ivankravets, shouldn't ```upload_protocol``` be honoured? Still not working even with the latest build. Can you please take a look?

PlatformIO version
```
Core [5.2.5]
Home [3.4.1]
```

VS Code version
```
Version: 1.65.2 (user setup)
Commit: c722ca6c7eed3d7987c0d5c3df5c45f6b15e77d1
Date: 2022-03-10T14:33:55.248Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044
```

commit 4b9eb5a8 (2019-03-28): Fix an issue when `upload_protocol` overriding does not work correctly // Resolve #7