platformio/platform-ststm8 #8: Build issues of example

rocketstrong600, 2019-03-28:
`PS C:\Users\12102376\Documents\PlatformIO\Projects\190328-172626-arduino-fade-all-pins> pio run
Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 12 compatible libraries
Scanning dependencies...
No dependencies
Archiving .pioenvs\stm8sblue\libFrameworkArduinoVariant.lib
Compiling .pioenvs\stm8sblue\FrameworkArduino\weak_serialEvent.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring_analog.c.rel*** [.pioenvs\stm8sblue\libFrameworkArduinoVariant.lib] Error 309

================================================================ [ERROR] Took 0.87 seconds ================================================================
PS C:\Users\12102376\Documents\PlatformIO\Projects\190328-172626-arduino-fade-all-pins>`

ivankravets, 2019-03-29:
Please provide your `platformio.ini`.

ivankravets, 2019-05-03:
Please reopen if you still need help

closed 2019-05-03 by ivankravets

ElliotCui97, 2019-05-13:
[env:stm8sblue]
platform = ststm8
board = stm8sblue
framework = arduino

ElliotCui97, 2019-05-13:
I have the same problem.

ElliotCui97, 2019-05-13:
> Executing task in folder STM8_LED_TEST: C:\Users\zjcui\.platformio\penv\Scripts\platformio.exe run --environment stm8sblue <

Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 12 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\stm8sblue\src\main.c.rel
Archiving .pioenvs\stm8sblue\libFrameworkArduinoVariant.lib
Compiling .pioenvs\stm8sblue\FrameworkArduino\HardwareSerial.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\Print-float.c.rel
*** [.pioenvs\stm8sblue\libFrameworkArduinoVariant.lib] Error 309
C:\Users\zjcui\.platformio\packages\framework-arduinoststm8\cores\sduino\HardwareSerial.c:38:2: warning: #warning
"using uart1 for HardwareSerial"

xfdr0805, 2019-06-02:
I have the same problem on win10 x64

Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 15 compatible libraries
Scanning dependencies...
No dependencies
Archiving .pioenvs\stm8sblue\libFrameworkArduinoVariant.lib
Compiling .pioenvs\stm8sblue\FrameworkArduino\weak_atexit.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\weak_initVariant.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\weak_serialEvent.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring_analog.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring_digital.c.rel
*** [.pioenvs\stm8sblue\libFrameworkArduinoVariant.lib] Error 309Compiling .pioenvs\stm8sblue\FrameworkArduino\wiring_pulse.c.rel

C:\users\xfdr0\.platformio\packages\framework-arduinoststm8\cores\sduino\weak_atexit.c:13: warning 85: in function atexit unreferenced function argument : 'func'
C:/users/xfdr0/.platformio/packages/framework-arduinoststm8/variants/standard/pins_arduino.h:260: warning 158: overflow in implicit constant conversion