platformio/platform-ststm8 #9: Please add board support for STM8L Discovery.

linuxforever, 2019-04-02:
Hi, 
I would like to request support for stm8l and its discovery development board. 
https://www.st.com/en/evaluation-tools/stm8l-discovery.html

The issue has been up before. https://github.com/platformio/platform-ststm8/issues/1

Thanks

LexNetAT, 2020-04-25:
So far only the STM8S/A series is apparently supported and none of the others - like STM8L. I think that's a shame because the repository was christened "STM8" - so all possible devices should be possible.
I'm already thinking about building it myself and hosting it on my GitHub for SPL.

umeshwalkar, 2020-06-16:
Hello LexNetAT,

hope your are well,
Did you manage to add STM8L series MCU and boards.
What it takes if I want to add L series MCU.

Thanks
Umesh

LexNetAT, 2020-06-16:
Hi Umesh,
I haven't found time to make any progress here yet. At that time I already gathered some information but the information is not available athoc. You are welcome to start working.

46cv8, 2023-11-06:
Add me to the list I would really love to have support for stm8l as well. I would love not to have to use the windows system under a MV. Is there anything potentially difficult about getting it working?