platformio/platform-ststm8 #12: *** [.pioenvs\stm8sblue\libFrameworkArduinoVariant.lib] Error 309

xfdr0805, 2019-06-10:
win10 x64

> Executing task: C:\Users\xfdr0\.platformio\penv\Scripts\platformio.exe run <

Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz 1KB RAM (8KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 15 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\stm8sblue\src\main.c.rel
Archiving .pioenvs\stm8sblue\libFrameworkArduinoVariant.lib
Compiling .pioenvs\stm8sblue\FrameworkArduino\HardwareSerial.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\Print-float.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\Print.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\SPI.c.rel
Compiling .pioenvs\stm8sblue\FrameworkArduino\WInterrupts.c.rel
*** [.pioenvs\stm8sblue\libFrameworkArduinoVariant.lib] Error 309
Compiling .pioenvs\stm8sblue\FrameworkArduino\WMath.c.rel
C:\users\xfdr0\.platformio\packages\framework-arduinoststm8\cores\sduino\WInterrupts.c:73: warning 85: in function attachInterrupt unreferenced function argument : 'mode'
C:\users\xfdr0\.platformio\packages\framework-arduinoststm8\cores\sduino\HardwareSerial.c:38:2: warning: #warning "using uart1 for HardwareSerial"
===================================================== [ERROR] Took 3.38 seconds =====================================================

hichemhlm, 2019-07-01:
Me too, any fix please ? Is it really possible to compile cpp in stm8s ?

xfdr0805, 2019-07-02:
> possible

main.cpp --->main.c  
in another pc can build success,but in my pc can not still build success,I have no idea too

odudex, 2019-09-24:
Having the same here.
Also tried spl platform without success, since blink LED example gave me 10KB of program size

cednik, 2020-05-14:
Me too - trying to build new empty project on Windows using PIO Core 4.3.3 in VSCode.
Renaming main.cpp ->main.c doesn't help.
And also compiler treat warnings as fatal error and stops compilation on the first warning ("using uart1 for HardwareSerial" from HardwareSerial.c in my case).

Santi-hr, 2020-11-08:
I'm having the same problems in PIO Core 5.0.2 in Visual Studio Code

Vampir8, 2020-11-10:
Me too.
```
PlatformIO Core             5.0.2
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1250
PlatformIO Core Directory   C:\Users\admin\.platformio
PlatformIO Core Executable  C:\Users\admin\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\admin\.platformio\penv\scripts\python.exe
Global Libraries            54
Development Platforms       5
Tools & Toolchains          31
```

maxgerhardt, 2020-11-10:
@valeros please help

HelloDB, 2021-01-17:
> @valeros请帮助

Use sdcc-4.0.0 to overwrite .platformio\packages\toolchain-sdcc, you can download sdcc-4.0.0 here->[https://sourceforge.net/projects/sdcc/files/](https://sourceforge.net/projects/sdcc/files/)
Also need to modify main.cpp->main.c, and add in the program: `void main(void);
void (*dummy_variable) () = main;`
main.c
```
#include <Arduino.h>

/* compilation fix. reference main function. */
void main(void);
void (*dummy_variable) () = main;

/* actual code.. */
void setup(){}
void loop(){}
```
platformio.ini
```
[env:stm8sblue]
platform = ststm8
board = stm8sblue
framework = arduino
```
I verified this method on windows10

[https://community.platformio.org/t/error-309-arduino-fade-all-pins-for-the-platform-st-stm8/13690/21](https://community.platformio.org/t/error-309-arduino-fade-all-pins-for-the-platform-st-stm8/13690/21)