platformio/platform-ststm8 #13: New SDCC version

ludiazv, 2019-09-13:
Some months ago SDCC team realeased a new version 3.9 that include serveral improvements for stm8 platform. Here are the release notes: https://sourceforge.net/p/sdcc/news/2019/04/small-device-c-compiler-390-released/

Pio now uses 3.8 version. I think it could be a good idea to update the platform with this new version.

Regerds,

maxgerhardt, 2021-02-02:
Update: There's now [SDCC v4.0](https://sourceforge.net/p/sdcc/news/2020/01/small-device-c-compiler-400-released/) and it even reduces firmware sizes in the SPL and Blinky firmwares :) 

Also some users reported that it doesn't work without the SDCC v4.0 toolchain (https://community.platformio.org/t/error-309-arduino-fade-all-pins-for-the-platform-st-stm8/13690/20?u=maxgerhardt).

nerocide, 2021-02-10:
I can confirm, i had to use sdcc 4.0 ;) thanks @maxgerhardt

gicking, 2021-03-06:
AFAIK the 4.1 release is already in release candidate status. Maybe you should wait a few more days...

gicking, 2021-03-08:
Update 2020-03-08: information from the SDCC project

The official release for **SDCC 4.1.0 is available** in our SourceForge File release system:
https://sourceforge.net/projects/sdcc/files/

In addition to the source package, binaries are available for 32- and 64-bit Windows, 64-bit macOS, and x86_64 GNU/Linux.

In addition to various bug fixes, notable features added since the 4.0.0 release are:
* New z80n port for the Spectrum Next CPU (a Z80 variant).
* Much better register allocation in the gbz80 backend.
* Workarounds for Rabbit wait state bugs in the r2k backend
* New r2ka port to better support Rabbit 2000A, 2000B, 2000C, 3000.
* Default crt0 and --data-loc for Rabbits suitable for typical Rabbit hardware configurations, such as the RCMs.
* Many improvements in code generation for z80 and related ports.
* Rabbit register definition headers for Rabbit 2000, 2000A, 2000B, 2000C, 3000, 3000A.
* C23 digit separators.

A full list of changes can be found in the ChangeLog:
https://sourceforge.net/p/sdcc/code/HEAD/tree/tags/sdcc-4.1.0/sdcc/ChangeLog

valeros, 2021-03-23:
Resolved in https://github.com/platformio/platform-ststm8/commit/4725bcd8908649cffcbe052e03a4ffaacc3d6764. Please retest with the upstream version:
```ini
[env:test]
platform = https://github.com/platformio/platform-ststm8.git
framework = ...
board = ...
```

closed 2021-03-23 by valeros