platformio/platform-ststm8 #16: Could not upload firmware using Stlink protocol

Qspace, 2020-04-25:
- Vscode version: 1.44.2
- Platform IO core: 4.3.1
- platformio.ini file: 
[env:stm8sdisco]
platform = ststm8
framework = spl
board = stm8sdisco
upload_protocol = stlink

[image]

Qspace, 2020-04-25:
When plugging USB cable to PC, board recognize as STM32
[image]

Qspace, 2020-04-28:
Can anyone help me solve this issue?

jxltom, 2020-09-30:
stlink is for stlinkv1, if you are using stlinkv2.1, please define ```upload_protocol = stlinkv2```

nerocide, 2021-02-09:
There is a lot of fixes to be done on stm8 sdcc etc.

I've found help here [stm8s-compiling-or-even-just-setting-up-to-work-vscode-help](https://community.platformio.org/t/stm8s-compiling-or-even-just-setting-up-to-work-vscode-help/18663/8)