platformio/platform-ststm8 #18: ASlink-Warning-Undefined Global '_assert_failed'

bogdan2011, 2020-05-28:
I get an error during the build process:
```
?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_adc1'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_awu'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_beep'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_clk'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_exti'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_flash'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_gpio'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_i2c'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_itc'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_iwdg'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_rst'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_spi'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_tim1'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_tim2'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_tim4'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_uart1'

?ASlink-Warning-Undefined Global '_assert_failed' referenced by module 'stm8s_wwdg'
*** [.pio/build/stm8sblue/firmware.elf] Error 1
```
And the build process fails completely. Might be due to the missing library (.lib) or .rel files not being linked properly.

boeckhoff, 2020-11-25:
take a look at the SPL examples, all of them define this method in main.c
`void assert_failed(uint8_t* file, uint32_t line)`

its needed to compile.

Rombersoft, 2025-11-30:
> take a look at the SPL examples, all of them define this method in main.c `void assert_failed(uint8_t* file, uint32_t line)`
> 
> its needed to compile.

It is wrong advice! In 2025, it does not help. I'm a beginner in PlatformIO, and I understand what garbage I have to deal with! I work with STM32CubeIDE and ArduinoIDE and have no problem. But I need an IDE for the STM8 MCU in Linux OS; therefore, I tried Platformio. So, I'm disappointed because I tried the first example with the UART, which was written 5 years ago, and it is not working. Support is missing, actual examples are missing, a new documentation is missing. My opinion is that this product is created by ancient people for ancient people. I'm Ukrainian too, but I do not support this project.