platformio/platform-ststm8 #21: New Project Buid Failed even after main.ccp -> main.c

piyushverma, 2020-06-01:
Hello,

I have renamed main.cpp to main.c it let move compilation move forware but at end it faile without any clear error message.
Here is log as below

platformio run -v
Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
-------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 1.0.2 > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3P6 16MHz, 1KB RAM, 8KB Flash
PACKAGES: 
 - framework-arduinoststm8 0.40.181216 
 - tool-stm8binutils 0.230.0 (2.30) 
 - toolchain-sdcc 1.30804.10766 (3.8.4)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 14 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
sdcc -o .pio/build/stm8sblue/firmware.elf -mstm8 --nostdlib --code-size 8192 --iram-size 1024 --out-fmt-elf .pio/build/stm8sblue/src/main.c.rel -L/home/piyush/.platformio/packages/toolchain-sdcc/share/sdcc/lib/stm8 -L/home/piyush/.platformio/packages/framework-arduinoststm8/STM8S_StdPeriph_Driver/lib .pio/build/stm8sblue/libFrameworkArduinoVariant.lib .pio/build/stm8sblue/libFrameworkArduino.lib -lstm8 -lSTM8S103
ASlink-Warning-No definition of area SSEG
*** [.pio/build/stm8sblue/firmware.elf] Error 1
============================ [FAILED] Took 0.72 seconds ============================

Any Suggestion ?

valeros, 2020-06-01:
Hi @piyushverma ! Are you able to compile your project using the same board in the Arduino IDE?

piyushverma, 2020-06-01:
yes I follow instruction from https://github.com/tenbaht/sduino I can compile blink project.
With more testing I notice, just by adding delay(1000); in loop function it build successfully.
hope this information can help others.

piyushverma, 2020-06-01:
yes I follow instruction from https://github.com/tenbaht/sduino I can compile blink project.
With more testing I notice, just by adding delay(1000); in loop function it build successfully.
hope this information can help others.

Any idea how delay is effecting build success/fail ?

mtbspace, 2020-06-12:
I've found the same here.  A new project will not build.  I can build if after completing the following steps:
1. Rename ```main.cpp``` to ```main.c```
2. Add ```delay(1000);``` to loop function as suggested by piyushverma

mtbspace, 2020-06-25:
Can anyone comment if we're doing something wrong, or does PlatformIO not work out the box with STM8 on some operating systems?

45gfg9, 2020-08-28:
This is weird, feels like something is broken

And, did anyone encounter problems with IntelliSense on header dependency (`stdlib.h` not found) problem?