platformio/platform-ststm8 #23: port tutorials to FOSS STM8 headers...?

gicking, 2020-07-14:
Hello,

I stumbed accross your STM8 examples on https://github.com/platformio/platform-ststm8 and would like to bring another project to your attention.

Unfortunately SDCC does not provide STM8 device headers, IAR and Cosmic each use a different syntax, and STM's SPL is not open-source. So quite often people create their own headers which are often incomplete and/or only fit a specific device - or use non-free headers from STM, IAR or Cosmic...

To overcome this and support the STM8 FOSS community, I am currently trying to establish "standard" FOSS STM8 device headers. You can find these, together with some examples, under https://github.com/gicking/STM8_headers. Current status is that the SDCC developer team is checking whether the headers should be bundled with SDCC.
These headers offer some gimmicks like optional bitfield access, but the main purpose is to establish a FOSS standard for STM8 which facilitates re-use of examples from different sources, as is common e.g. for AVR or PIC. To increase acceptance, I also provide some examples with Makefile (SDCC) and projects for Cosmic & IAR, respectively in that repository.

If you would like to support the idea of standard FOSS STM8 device headers, please consider porting your tutorials to these new headers - and spread the word. You will see from the examples in the repo that this is rather straightforward.

For your support thanks a lot in advance! Have a great day  :-)

Regards,
Georg Icking-Konert (gicking)

valeros, 2021-03-23:
Hi @gicking ! There is a great PR pending in your repo https://github.com/gicking/STM8_headers/pull/5 that will greatly simplify using the FOSS STM8 headers within the ST STM8 platform (#46). Please consider merging that library manifest.

gicking, 2021-03-24:
hi @valeros Thanks a lot for your great effort with PIO! I am already wondering why I bothered so long with Makefiles...

Just merged your pull-request [gicking/STM8_headers#5](https://github.com/gicking/STM8_headers/pull/5)

valeros, 2021-03-25:
Resolved in #46

closed 2021-03-25 by valeros