platformio/platform-ststm8 #25: STM8S_BLUE c_cpp_properties.json wrong F_CPU

ipa64, 2020-10-26:
Hi,
By default F_CPU for STM8S_BLUE "STM8S103" is wrong in the c_cpp_properties.json file 
"F_CPU=160000000L" (160MHz) instead of "F_CPU=16000000L" (16MHz)

Thank you

ipa64, 2020-10-26:
Hi, that was my mistake in fact c_cpp_properties.json is build from platformio.ini

maxgerhardt, 2021-01-20:
It's defined in 

https://github.com/platformio/platform-ststm8/blob/cc3da36621bc32e1df23cc110963daeb6e1b187c/boards/stm8sblue.json#L5

And according to datasheet (https://www.st.com/en/microcontrollers-microprocessors/stm8s103f3.html) that's the maximum frequency. If a project / board doesn't have an HSE, users can always redefine it in the `platformio.ini` with

```
board_build.f_cpu = 2000000
```
just like [docs](https://docs.platformio.org/en/latest/boards/ststm8/stm8sblue.html#configuration) show

valeros, 2021-03-23:
The `f_cpu` field contains the max frequency. Please edit this value if you need as @maxgerhardt suggested.

closed 2021-03-23 by valeros