platformio/platform-ststm8 #27: SyntaxError `EOL while scanning string literal

ondraco, 2021-01-29:
Hello, 

I am trying to use the platformio for STM 8 but on compilation i keep getting this error:

```
> Executing task in folder STM8s103TEST: pio run <
Processing stm8sblue (platform: ststm8; board: stm8sblue; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sblue.html
PLATFORM: ST STM8 (1.0.2) > ST STM8S103F3 Breakout Board
HARDWARE: STM8S103F3 16MHz, 1KB RAM, 8KB Flash
PACKAGES: 
 - framework-arduinoststm8 0.40.181216 
 - tool-stm8binutils 0.230.0 (2.30) 
 - toolchain-sdcc 1.30804.10766 (3.8.4)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
*** [.pio/build/stm8sblue/firmware.ihx] SyntaxError `EOL while scanning string literal (<string>, line 1)' trying to evaluate `${TEMPFILE('$LINK -o $TARGET ${__ldflags_for_hex(__env__, LINKFLAGS)}'
======================================================================================================================== [FAILED] Took 0.30 seconds ========================================================================================================================
The terminal process "pio 'run'" terminated with exit code: 1.
```

platformio.ini:
```
[env:stm8sblue]
platform = ststm8
board = stm8sblue
framework = arduino

upload_protocol = serial

; change microcontroller
board_build.mcu = stm8s103f3

; change MCU frequency
board_build.f_cpu = 16000000L
```

I tried on on both windows and Linux. I even tried a completely fresh installation of platformio on different computer with the STM 8 examples its always the same error. The compilation and deployment works fine in Arduino IDE. The STM 32 works without issue in platformio.

Thank you for any help!

Regards,
Ondrej

maxgerhardt, 2021-01-31:
Also reported in https://community.platformio.org/t/issue-building-stm8/18966.  This seems to be a problem in the platform  script or / and in conjuction with the new SCons versions used in the core. 

@valeros

closed 2021-02-01 by valeros

valeros, 2021-02-01:
Should be fixed in the latest platform version `1.0.3`, please update via `pio platform update ststm8`.

maxgerhardt, 2021-02-01:
Confirmed working after update.

```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm8/mb208.html
PLATFORM: ST STM8 (1.0.3) > sduino MB (STM8S208MBT6B)
HARDWARE: STM8S208MBT6 16MHz, 6KB RAM, 128KB Flash   
PACKAGES:
..
RAM:   [          ]   0.3% (used 21 bytes from 6144 bytes)
Flash: [==        ]  22.2% (used 29062 bytes from 131072 bytes)
=================== [SUCCESS] Took 4.77 seconds ===================
```

analityk, 2021-02-02:
> [2.02.2021, 09:42:26] Unable to resolve configuration using compilerPath property" C: /Users/Szymon/.platformio/packages/toolchain-sdcc/bin/sdcc.exe ". Instead, the value "cl.exe" will be used. 

What does it mean?

maxgerhardt, 2021-02-02:
>What does it mean?

It means that VSCode's C++ extension can't handle SDCC.exe as the compiler. Supports only clang, llvm, gcc and Microsoft's compiler (for e.g. Visual Studio). Related to https://github.com/platformio/platform-ststm8/issues/26, but this has nothing to do with the issue that was being discussed in this issue page.

commit 156889a9 (2021-02-01): Fix compatibility with the PIO Core 5.1 // Resolve #27