platformio/platform-ststm8 #29: STM8 L,A platform support

AttilaDanyadi, 2021-02-11:
STM8L and and STM8A devices are not supported. These devices have completely different StdPeriph libs than STML8S
Would be fancy if we should not need to define "demo board"

CSharperMantle, 2023-01-30:
Any progress on this?

I noticed that almost all STM8 MCUs have been listed as NRND on ST website. Does that has any negative impact on this issue?

46cv8, 2023-11-05:
I see that some of the STM8A boards seem like they might be supported but not the STM8L boards. Is there something that makes this difficult or is it simply a matter of adding afew template files for a board I wish to program for? Do I need to create/install a custom version of OpenOCD with additiona lchanges to support my chip? STM8L101F3