platformio/platform-ststm8 #34: Please add support for Nucleo-8S207K8 board

gicking, 2021-03-06:
please consider adding support for the Nucleo-8S207K8 development board by STM. Thanks a lot in advance!

https://www.st.com/en/evaluation-tools/nucleo-8s207k8.html

maxgerhardt, 2021-03-06:
Hey, do you have the board to test? 

Sadly the board [isn't supported](https://github.com/tenbaht/sduino/blob/development/sduino/stm8/boards.txt) by the underlying SDuino Arduino core implementation (for 8S200 series, only a S208MB board is supported), but SPL should still work. 

To test SPL support, please copy the project files from https://github.com/platformio/platform-ststm8/tree/develop/examples/spl-blink and 
1. Create a new folder `boards` in the root of the project 
2. Create a new file `nucleo_8s207k8.json` in there
3. Fill it with the content
```json
{
    "build": {
      "core": "sduino",
      "extra_flags": "-DSTM8S_207K8 -DSTM8S207",
      "f_cpu": "16000000L",
      "cpu": "stm8",
      "mcu": "stm8s207k8t6",
      "variant": "mb208"
    }, 
    "frameworks": [
      "arduino",
      "spl"
    ],
    "upload": {
      "maximum_ram_size": 6144,
      "maximum_size": 65536,
      "protocol": "stlinkv2",
      "protocols": [
        "stlinkv2",
        "serial"
      ]
    },
    "name": "NUCLEO-8S207K8",
    "url": "https://www.st.com/en/evaluation-tools/nucleo-8s207k8.html",
    "vendor": "STMicroelectronics"
  }
```
4. Change the `platformio.ini` of the project to 
```ini
[env:nucleo_8s207k8]
platform = ststm8
board = nucleo_8s207k8
framework = spl
upload_protocol = stlinkv2
upload_command = $PROJECT_PACKAGES_DIR/tool-stm8tools/stm8flash -p stm8s207k8 -c $UPLOAD_PROTOCOL -s flash -w $SOURCE
```
5. In `src/main.c`, adapt the GPIO pins and ports to
```c
#define LED_GPIO_PORT  (GPIOC)
#define LED_GPIO_PINS  (GPIO_PIN_5)
```
(since the green LD3 [is connected there](https://www.st.com/resource/en/user_manual/dm00489875-stm8-nucleo32-board-mb1442-stmicroelectronics.pdf))
6. Try to compile and upload the firmware

Build should result in 

```
Building in release mode
Compiling .pio\build\nucleo_8s207k8\SPL\stm8s_adc2.rel
..
Compiling .pio\build\nucleo_8s207k8\SPL\stm8s_wwdg.rel
Compiling .pio\build\nucleo_8s207k8\src\main.rel
Compiling .pio\build\nucleo_8s207k8\src\stm8s_it.rel
src\main.c:114: warning 85: in function assert_failed unreferenced function argument : 'file'
src\main.c:114: warning 85: in function assert_failed unreferenced function argument : 'line'
C:\Users\Max\.platformio\packages\framework-ststm8spl\Libraries\STM8S_StdPeriph_Driver\src\stm8s_tim2.c:1110: warning 116: right shifting more than size of object changed to zero
C:\Users\Max\.platformio\packages\framework-ststm8spl\Libraries\STM8S_StdPeriph_Driver\src\stm8s_itc.c:60: warning 59: function 'ITC_GetCPUCC' must return value
Linking .pio\build\nucleo_8s207k8\firmware.elf
Checking size .pio\build\nucleo_8s207k8\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 5 bytes from 6144 bytes)
Flash: [====      ]  37.9% (used 24837 bytes from 65536 bytes)
======================= [SUCCESS] Took 2.99 seconds =======================
```
And upload should trigger a 

```
C:\Users\<user>\.platformio\packages/tool-stm8tools/stm8flash -p stm8s207k8 -c stlinkv21 -s flash -w .pio\build\nucleo_8s207k8\firmware.ihx
```
..which hopefully works. I don't have the hardware to test.

gicking, 2021-03-07:
:-)

worked right out of the box. Thanks a lot! Will this be added to the list of boards now - though "only" with SPL support? 

If yes, would you please also consider adding the other [Nucleo-8S207K8](https://www.st.com/en/evaluation-tools/nucleo-8s207k8.html) board...? Thanks in advance!

maxgerhardt, 2021-03-07:
>worked right out of the box. 

Great to hear that it works (and the LED seem to be blinking yeah?).

>If yes, would you please also consider adding the other Nucleo-8S207K8 board...? Thanks in advance!

But this alsready the 82S207K8 board, I think you mean #33? Yep, shouldn't be hard to have basic SPL supoprt there to, again sadly no Arduino integration (yet). 

Though I might also look into that quickly, because all a [variant](https://github.com/tenbaht/sduino/tree/development/sduino/stm8/variants) is is one header file defining the Arduino pin mapping, standard pins and some info regarding timers. Thus maybe also a different variant works that has 32-pin package (?). Not sure about that and without hardware to test dangerous to test fully and PR, so I might just get one of these boards.

>PS: or is the right way to clone the repo, add the changes and create a pull request...?

Exactly, I'd have to cleanup the board JSON file(s), modify some python code so that the `upload_command` directive isn't needed (only needed because PlatformIO has a [wrong derivation logic](https://github.com/platformio/platform-ststm8/blob/5bb1a69d699e33e96d6f9a4351e2510975bfc90f/builder/main.py#L155) the `-p` switch in that case) and then PR it into here, then the devs should merge it.

valeros, 2021-03-23:
Resolved in #38

closed 2021-03-23 by valeros