platformio/platform-ststm8 #40: Ability to change RAM values displayed via the MEMORY view?

IansMusical, 2021-03-23:
@maxgerhardt I've been speaking with Georg who recently informed me of this project. Thank you for the work you are doing it looks great so far!

In the MEMORY view of PlatformIO it is possible to display in a new window the STM8x RAM values. However, I've not found a way to change a RAM value. Could this feature be added please?

maxgerhardt, 2021-03-23:
Hey @IansMusical, I think this is feature that must be generally added in the debugger view, I think that code lives in https://github.com/platformio/platformio-vscode-ide/. 

For now it should be possibly however with direct GDB commands, as e.g.

```
set {int}0x0004 = 4
```
writes 4 as an integer-value (I think 2 bytes here?) at address 0x0004.  

@valeros this issue should be transfered I think?

IansMusical, 2021-03-24:
@maxgerhardt Thank you for your reply. I have successfully been able to use GDB directly as per your instructions to set a global variable, which updated in the VARIABLES view. I was also able to update core registers, e.g. a, xl and view them in GDB but the REGISTERS view did not automatically update even if I closed and opened the view again.

Will you be the person to connect these GDB commands to the GUI (somehow) or is it in the core?

maxgerhardt, 2021-03-25:
>Will you be the person to connect these GDB commands to the GUI (somehow) or is it in the core?

Per https://github.com/platformio/platform-ststm8/issues/42#issuecomment-807565708