platformio/platform-ststm8 #42: Ability to see disassembly and step assembly code?

IansMusical, 2021-03-23:
@maxgerhardt: Is it possible to see STM8 disassembly in a window and equally be able to step over assembly code?

maxgerhardt, 2021-03-23:
This is working in e.g. STM32 platforms, just as described in the featureset at https://docs.platformio.org/en/latest/integration/ide/vscode.html#debugging and per screenshot at https://docs.platformio.org/en/latest/integration/ide/pioide.html#platformio-for-vscode, but somehow this is not working for STM8 yet (and also [not Atmel AVR](https://github.com/platformio/platform-atmelavr/issues/230)), I'll open issues and try and see why exactly this is not working in this instance.

IansMusical, 2021-03-24:
@maxgerhardt Thank you for your reply. Does this feature work at a core level as I couldn't immediately see anything in the ststm32 repo.? Again, will this be something you can change once the issue is found or is it someone else's change?

maxgerhardt, 2021-03-25:
>Does this feature work at a core level as I couldn't immediately see anything in the ststm32 repo.? 

This is not implemented in the PlatformIO core (https://github.com/platformio/platformio-core/) but the VSCode extension (https://github.com/platformio/platformio-vscode-ide/ and its dependencies). 

>Again, will this be something you can change once the issue is found or is it someone else's change?

PlatformIO staff will be in charge of that I think, because I don't see the public source for some of the debugger GUI stuff (I might also be wrong). After all, I'm just a contributer, not an employee :D