platformio/platform-ststm8 #52: tool-stm8binutils was linked against ancient shared libraries libmpfr4 and libgmp3

retiredfeline, 2021-09-22:
I think you are not the maintainers of this support package, but I'd like to point out that it was compiled some time ago and is probably out of date. When I tried to start a debug session it said that stm8-gdb could not find the shared library libmpfr4. A ldd of the binary showed:

> stm8-gdb:
>         linux-vdso.so.1 (0x00007ffe89181000)
>         libdl.so.2 => /lib64/libdl.so.2 (0x00007f7eb4e2e000)
>         libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f7eb4c0e000)
>         libm.so.6 => /lib64/libm.so.6 (0x00007f7eb48cd000)
>         libpython2.7.so.1.0 => /usr/lib64/libpython2.7.so.1.0 (0x00007f7eb4489000)
>         libexpat.so.1 => /usr/lib64/libexpat.so.1 (0x00007f7eb4257000)
>         libmpfr.so.4 => not found
>         libc.so.6 => /lib64/libc.so.6 (0x00007f7eb3e82000)
>         /lib64/ld-linux-x86-64.so.2 (0x00007f7eb5032000)
>         libutil.so.1 => /lib64/libutil.so.1 (0x00007f7eb3c7f000)

A search for this library and the dependency libgmp3 for my OS (openSUSE 15.3) only found packages from older releases. Fortunately I was able to install those RPM package manually because they didn't clash with the current packages libmpfr6 and libgmp10 (that's how old the shared libraries used by stm8-gdb are) and had no other dependencies. After that my debug session in VSCode worked. Users of other Linux distros may experience a similar problem.

And looking at the dependency list there may be a looming problem with libpython2.

fabalthazar, 2021-12-13:
Same problem for me. I am stuck with unexecutable stm8-gdb on a xubuntu 20.10 machine.
I will try to build it from source.

fabalthazar, 2022-01-23:
Upstream stm8-binutils-gdb has been updated since the last PlatformIO package.
Manually building it from source and specifying it in platformio.ini has shown success on my side.

The last upstream version dates back to July 18th, 2021: stm8-binutils-gdb-sources-2021-07-18.tar.gz.

https://sourceforge.net/projects/stm8-binutils-gdb/files/

Is it possible to update the tool-stm8binutils pio package?
Thanks