platformio/platform-ststm8 #57: Unable to start debug session

ViggoAvatar, 2022-07-28:


gicking, 2022-10-28:
To get STM8 debugging via STLink to work under Ubuntu 22.04, I had to perform the below 2 steps:

1. install missing libpython2.7.so.1.0 --> sudo apt-get install libpython2.7
2. create symbolic link for libmpfr.so.4 --> sudo ln -s /usr/lib/x86_64-linux-gnu/libmpfr.so.6 /usr/lib/x86_64-linux-gnu/libmpfr.so.4

I found the hints [here](https://stackoverflow.com/questions/20842732/libpython2-7-so-1-0-cannot-open-shared-object-file-no-such-file-or-directory) and [here](https://stackoverflow.com/questions/51288467/cannot-open-libmpfr-so-4-after-update-on-ubuntu-18-04). Good luck!

reimerp, 2022-12-11:
Same issue here, but its not related to starting a debug seesion, but to compile in debug mode.
To actual start stm8-gdb above two actions are required unless stm8-gdb won't start, but the coredump is when "--debug" is issued. I am currently not sure if this is related to sduino lib / sdcc or some memory issues. I can compile WInterrupts.c manually without --debug. but with debug the core dumps.