platformio/platform-ststm8 #61: Debug fails: Failed to load symbols from executable file: Error: Could not find "objdump" program

lunamera, 2023-09-10:
### Configuration

**Operating system**:
Linux x86

**PlatformIO Version** (`platformio --version`):
pio --version
PlatformIO Core, version 6.1.11

I've read the release note: "Resolved a possible issue that may cause generated projects for [PlatformIO IDE for VSCode](https://docs.platformio.org/en/latest/integration/ide/vscode.html) to fail to launch a debug session because of a missing “objdump” binary when GDB is not part of the toolchain package"

STM8 target.

### Description of problem

STM8 debugging fails: "Failed to load symbols from executable file: Error: Could not find "objdump" program"

#### Steps to Reproduce

1. setup STM8 sample project:

```
[env:genericf207]
platform = ststm8
board = genericf207
framework = spl
debug_tool = stlink
upload_protocol = stlinkv2

```
Any minimal sample program is sufficient for reproduction:
```
int main ()
{
  while(1);
}

```

2. upload the code, make sure debugger connection is OK
3. start debugging  

### Actual Results

Error message shows up:  "Failed to load symbols from executable file: Error: Could not find "objdump" program

### Expected Results

Debugging should work

### Additional info

Content of launch.json:
```
// AUTOMATICALLY GENERATED FILE. PLEASE DO NOT MODIFY IT MANUALLY
//
// PIO Unified Debugger
//
// Documentation: https://docs.platformio.org/page/plus/debugging.html
// Configuration: https://docs.platformio.org/page/projectconf/section_env_debug.html

{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            "executable": "/home/lunamera/Documents/PlatformIO/Projects/stm8debugtest/.pio/build/genericf207/firmware.elf",
            "projectEnvName": "genericf207",
            "toolchainBinDir": "/home/lunamera/.platformio/packages/toolchain-sdcc/bin",
            "internalConsoleOptions": "openOnSessionStart",
            "svdPath": "/home/lunamera/.platformio/platforms/ststm8/misc/svd/STM8S207CB.svd",
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            }
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (skip Pre-Debug)",
            "executable": "/home/lunamera/Documents/PlatformIO/Projects/stm8debugtest/.pio/build/genericf207/firmware.elf",
            "projectEnvName": "genericf207",
            "toolchainBinDir": "/home/lunamera/.platformio/packages/toolchain-sdcc/bin",
            "internalConsoleOptions": "openOnSessionStart",
            "svdPath": "/home/lunamera/.platformio/platforms/ststm8/misc/svd/STM8S207CB.svd"
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (without uploading)",
            "executable": "/home/lunamera/Documents/PlatformIO/Projects/stm8debugtest/.pio/build/genericf207/firmware.elf",
            "projectEnvName": "genericf207",
            "toolchainBinDir": "/home/lunamera/.platformio/packages/toolchain-sdcc/bin",
            "internalConsoleOptions": "openOnSessionStart",
            "svdPath": "/home/lunamera/.platformio/platforms/ststm8/misc/svd/STM8S207CB.svd",
            "loadMode": "manual"
        }
    ]
}

```

valeros, 2023-09-12:
Hi @lunamera, thanks for reporting. Indeed the new PlatformIO core v6.11 contains changes that may cause this issue. Until we figure out how to solve it, could you please temporarily copy the `stm8-objdump` binary from `/home/YOUR_USER/.platformio/packages/tool-stm8binutils/bin` to `/home/YOUR_USER/.platformio/packages/toolchain-sdcc/bin` and try again?

lunamera, 2023-09-17:
Hi @valeros , I've copied the file & debugging is working now!
Thanks for the workaround and sorry for filing the bug against the wrong repo.